OPEnSLab-OSU/Nanolux #2: ESP32 Arduino IDE Driver issues OSX Monterey 12.3

gn1m1ew, 2022-04-18:


gn1m1ew, 2022-04-18:
Error Message:
exec: "python": executable file not found in $PATH
Error compiling for board Adafruit ESP32 Feather.

gn1m1ew, 2022-04-18:
To fix this issue:
1. In Arduino preferences, select and enable Show verbose output during compilation and upload then compile again.
2. Notice the entire command where it is failing and you must see it will probably be failing at python .../gen_esp32part.py ...
3. Edit the board compiler's platform config using the command open $HOME/Library/Arduino15/packages/esp32/hardware/esp32/2.0.2/platform.txt
4. Search for gen_esp32part.py and replace that python with entire path of your python. Like below:
# From
tools.gen_esp32part.cmd=**python** "{runtime.platform.path}/tools/gen_esp32part.py"
# To
tools.gen_esp32part.cmd=**/usr/local/bin/python** "{runtime.platform.path}/tools/gen_esp32part.py"
5. Save and close the file.
6. Try compile the Arduino file again.

goldswol, 2023-05-19:
Our team found no instances of this issue during development

closed 2023-05-19 by goldswol